christianhaitian/arkos #983: r33s arkos shutdown sequence starts up arkos boot animation and hangs.

canyoubarrett, 2024-03-10:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

 arkos menu shutdown sequence starts up arkos boot animation and hangs. doesnt shut down.

this must be my fault because it was working prior to editing a service file. 

I edited the emulationstation.service file to load an app, and upon quit, load emulationstation.

not sure why thats affecting the shutdown process. 

### How can the issue be reproduced? (If applicable)

i  added an image showing the file i edited and added. the emu service file boots up m8c and upon exit, loads the m8 service file i made which was a copy of the original emustation service file. 
[image]

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

r33s with the newest ark image for the r3xs

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

https://aeolusux.github.io/ArkOS-R3XS/ wont update any further.

### Anything else you'd like to include that may help to review this issue or feature request?

i know this is outside your scope of support. but i wanted to try and see if maybe you might know what the hang up is from. 

my goal was to boot into m8c app, the  close and auto open emulationstation on m8c exit. it all works great, but when i do the shutdown command, itll hang on the arkos boot animation. 

any pointers are appreciated, but i also understand if you have broader issues to address elsewhere and i am now reading, that this device is not supported. welp, but if anything comes to mind that I can poke at, please let me know. 

much appreciated, arkos is hands down the best OS for these handhelds. Ill be fine if you dont have an idea off hand or solution. this is a niche new issue that wont affect anyone else i imagine

christianhaitian, 2024-03-10:
First off, don't edit LInux files directly via a Windows machine.  That will screw with permissions.  second, after you created this service file, did you do a `sudo systemctl daemon-reload` so systemd would know about it?

canyoubarrett, 2024-03-10:
thanks for getting back so quickly. i did reload the service i created with that command

yeah windows is a nightmare to do this with. I wont use anymore. i didnt know about permission issues.

could permission issues bork the shutdown sequence? If i cant figure this out, ill start over without windows and see whats up. 

is there a shutdown file somewhere that i can look at? im just not sure why shutting down would break.

everything works though except that.

canyoubarrett, 2024-03-10:
i can or you can close this ticket out if its too far out of scope and support

closed 2024-03-10 by canyoubarrett

canyoubarrett, 2024-03-10:
any ideas?

christianhaitian, 2024-03-10:
no unfortunately

canyoubarrett, 2024-03-10:
that’s okay, thank you