christianhaitian/arkos #986: Gamecube emulation

dani7959, 2024-03-12:
.rvz files seem to work so they should be included in es_systems.cfg

Also, a couple of questions:
- tthere is no documented hotkey for the dolphin menu. is there none?
- there is no documentation for adding the gamecube bios. could it improve perfromance?

Getting 1-10 fps in 3d games.

christianhaitian, 2024-03-12:
I'll add this extension as a supported formatted in this month's update.

I don't know what your device is but available hotkeys for the dolphin emulator are listed in the faq for each supported rk3566 device.  For example, for the rg353v, see here: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353v#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS under the **For Dolphin standalone emulator** section

The dolphin emulator available here does not support loading a bios as far as I know.

christianhaitian, 2024-03-12:
Also, Gamecube emulation on this chipset is more of a proof of concept versus actually usable.

christianhaitian, 2024-03-29:
I forgot to accommodate this in this month's update.  I'll add it next month instead.

christianhaitian, 2024-03-30:
I've added .rvz support with today's update.

closed 2024-03-30 by christianhaitian