christianhaitian/arkos #987: No audio problem in ES

lauren7ino, 2024-03-12:
Hi

Yesterday after hours of fiddling with my RG353M with the "no audio" problem, I gave up and reinstalled ArkOS from the ground up. That fixed the problem.

Today I hand restored this ES setting and the "no audio" problem was also restored.
Can you test it and see if there is some lingering bug?
Too many people report no audio problem in ArkOS and almost none can fix it without reinstalling from scratch. I'm hoping this file helps.

Thanks

[es_settings.cfg.zip](https://github.com/christianhaitian/arkos/files/14577369/es_settings.cfg.zip)

christianhaitian, 2024-03-12:
If this file is from the backup process that is within ArkOS, that does not backup any files that would have any type of impact on audio.  There is a bug in the kernel that will cause that issue sometimes.  Usually, a reboot fixes it if you're on the latest version of arkos (02232024).  If not, here are a few other items you can try:

  - Press L3+R3 to restore audio.
  - Plug a 3.5mm pair of headphones then unplug them.  
  - Lastly, restoring the files from here for your model: https://github.com/christianhaitian/arkos/issues/772#issuecomment-1732564970 will fix the issue.  
 
I constantly shutdown and boot my install of ArkOS and I've only ran into the issue once over hundreds of boots and a simple reboot fixed the issue for me.

closed 2024-03-12 by christianhaitian

lauren7ino, 2024-03-13:
Are you sure there aren't two problems contributing to "no audio" problem?

Jut to say: I tried everything, rebooting, plugging a jack, etc., and nothing worked.
I noticed that in ES settings, the volume always went down to 0 after setting it to other value.
So I thought ES was the culprit.

christianhaitian, 2024-03-13:
Sorry for the rg353m, it's F+l3.  You can find that information in the faq here: <https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353m#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS>

No settings backed up in the process would cause that issue you described.  Either the device was accidently muted or the .asoundrc.cfg file located in `/home/ark` got corrupted.  The corruption of the .asoundrc.cfg file can happen from using a usb dac or bluetooth audio but usually a reboot of the device should solve that issue.