christianhaitian/arkos #988: PortMaster update, creates 2 PortMaster entries in Options menu

lauren7ino, 2024-03-12:
I had one entry in the Options menu with PortMaster.
I auto-updated PortMaster, and now the Options menu has 2 entries.

   * PORTMASTER
      * CONFIG
         PORTMASTER
   PORTMASTER

christianhaitian, 2024-03-12:
I'm no longer involved with maintaining portmaster.  You can report the issue here: https://github.com/PortsMaster/PortMaster-New

closed 2024-03-12 by christianhaitian

lauren7ino, 2024-03-15:
I meant in the Emulation Station Options menu.

christianhaitian, 2024-03-15:
Still caused by an issue with the portmaster update.

lauren7ino, 2024-03-20:
Seems to be fixed: https://github.com/PortsMaster/PortMaster-New/issues/193#issuecomment-2009211931
Thanks

christianhaitian, 2024-03-20:
Yes and I will include this fix in the next update as well in case others run into this issue.  Thanks for reporting it to the portmaster team and helping to test the fix.