christianhaitian/arkos #989: RG353V PPSSPP Cheat.db Not Show And Genesis/Megadrive 6 Button Problem

jericho82, 2024-03-13:
I have been download and upgrade ArkOS latest in my RG353V but Ppsspp cheat.db never show, i put on ppsspp/psp/cheats/ in everywhere include in easy rom, storage/my 2nd card and even in .config folderi pasting ppsspp folder i put cheat db but nothing show and got warning CANT CREATE CHEAT FILE, DISK MAYBE FULL or something like that, please fix it and the 2nd problem is genesis/megadrive Mortal Kombat 1 roms it seems whatever i change input controller into MD 6 Button it never work even i change the core genesis plus gx, wide and picodrive, please fix this too, i really love ArkOS so much, thanks in advance

jericho82, 2024-03-13:
First, that video is not help at all, i do more than the guy do, and i support ppsspp gold too so i know where the cheat.db belong folder BUT in ArkOS lates it not showing at all, different at base/default os and JelOs all fine and show, and the genesis problem i do like yours, changing into MD 6 button even changing into MD 6 button + 4 ways or something but nothing change and the characters it seems plautomatically pressing and hold Block/B/Y button

christianhaitian, 2024-03-13:
For the MK1 issue, maybe try a different rom because again, I tested this on my end following the same instructions and it worked fine.

For the cheats.db file, I don't know what's going on with your setup but I just tested loading this cheats.db file from [here](https://github.com/Saramagrean/CWCheat-Database-Plus-/tree/master) and it loaded just fine.  I have a 2 sd card setup on my RG353M and I copied that file to the psp\ppsspp\PSP\Cheats folder, then loaded God Of War, then loaded the game with standalone ppsspp, then hit R3, Then B on Cheats, then B on Import cheats from psp/ppsspp/PSP/Cheats and cheats showed up fine.

jericho82, 2024-03-13:
Pardon me dev, hmm maybe you know like a chocholate strawberry and chocolate banana, it has same specs but IT still different things, like RG 353V/M/P/VS it has maybe same specs BUT it still different device, on some forum someone has success cheat.db via EASYROM ppsspp folder then psp/cheats folder and now the dev itself told me to put usually at sdcard ppsspp/psp/cheat folder, BTW dev that cheat its make by me because i am the cheat maker at some bigger forum at past including like Codemaster Project, GameHacking, even i created for the first time 3DS gateway cheat codes, so it impossible i make wrong format cheat database for playstation portable, hmm okay then it not solve but forcing closed... I just download again ArkOs but i never update hope it will work if its no then i have no choose but to back into original linux os, i love the ArkOs but i do need psp cheats for getting some 60fps on some game THAT never have been found on internet brows because i made em, sorry dev i let my emo little carry on but trust me its same like PSP 100x and 200x and 300x, it maybe HAS same specs BUT it really make a different performa even its only little

christianhaitian, 2024-03-13:
I have no idea what you're saying here but good luck and I hope you find a solution for your needs.

jericho82, 2024-03-14:
> I have no idea what you're saying here but good luck and I hope you find a solution for your needs.

It seems ArkOS PsP in RG353V cannot using CHEAT.DB but using other cheat method IDNUMBER.ini file, but NOT detected while Cheat.db still in CHEATS foldet so Cheat.db must be deleted from CHEATS folder