christianhaitian/arkos #990: Is it possible to manually update and/or replace components?

fexofenadine, 2024-03-15:
I noticed that due to the organization of the file structure, certain things are difficult to modify, in the usual "Ubuntu"-like way.
For example, using apt to install components often fails to complete (I think due to symlinking).  Retroarch cannot be manually reinstalled the usual way as it's in a different folder structure than is usual.

I came across the issue as I was interesting in using retroarch on its won in ArkOS, ie, liad Retroarch , generate some playlists, run it all from in thee.  The version that ships with ArkOS seems to be modified from stock, I was hoping to be able to manually reinstall the stock version for those Content-Scan generated playlists and so on

Anything you can do to help would be great.  Of course a lot of what I describe seems to be by design, but having a more standardized Ubuntu backend would provide obvious development opportunities.  If theres a wiki that could help me do what I'm after, that would be helpful too.

Thanks so much, I love this platform

christianhaitian, 2024-03-15:
It is possible to install apt packages.  Just be sure to be on the latest version of ArkOS( (02232024) before trying this.  This is a very customized install so there are some changes from the ubuntu/debian way customized for portable gaming.  If you would like to run Retroarch directly, you can do that from the BaRT menu. Check the FAQ available through the wiki for your particular device for info on accessing BaRT.

closed 2024-03-15 by christianhaitian

fexofenadine, 2024-03-15:
this is great, thanks for your response, Christian