christianhaitian/arkos #992: Battery icon text issue

tanimz2, 2024-03-15:
The text for the battery icon is too far to the right and not visible. Theme - ARKOS CARBON. Device - R351MP
[image]

christianhaitian, 2024-03-15:
Make sure to the theme configuration correctly to 4:3

closed 2024-03-15 by christianhaitian

mdeguzis, 2024-05-28:
Many themes don't have that option or have a "optimize for small screens" option that does little (like ES-THEME-ARKOS-CARBON). I have that same issue on the theme. It really is too far right in most cases and only needs to go over a smidge.  I guess we have to work with the theme makers then. Didn't even ask which theme he was using :)

mdeguzis, 2024-05-28:
With the above change here is at 100%:

[image]