christianhaitian/arkos #994: Update: stuck on 'You didn't type OK'

wouterVE, 2024-03-16:
### Describe the Issue (If applicable)
When entering the update script you first need to enter OK in order to proceed. However, no matter what I type (ok/OK) and exit the screen with the function button + exit (II+VI) I always get the following notification:

`You didn't type OK.  This script will exit now and no changes have been made from this process.`

I already tried to hack the update script (`ArkOKUpdate.sh`) by doing the following things:

- Add a the following line:

```
sudo msgbox "ONCE YOU PROCEED WITH THIS UPDATE SCRIPT, DO NOT STOP THIS SCRIPT UNTIL IT IS COMPLETED OR THIS DISTRIBUTION MAY BE LEFT IN A STATE OF >
my_var=`osk "Enter OK here to proceed." | tail -n 1`
#My code
my_var="OK"

echo "$my_var" | tee -a "$LOG_FILE"

if [ "$my_var" != "OK" ] && [ "$my_var" != "ok" ]; then
  sudo msgbox "You didn't type OK.  This script will exit now and no changes have been made from this process."
  printf "You didn't type OK.  This script will exit now and no changes have been made from this process." | tee -a "$LOG_FILE"
  exit 187
fi
```
However, upon starting the update process it seems that my changes are reverted.

- Next I tried to type manually OK in the associated log (`update02232024.log`) but this also didn't affect anything.

So it seems the keyboard is not saving the variable OK  correctly - also not when I'm hacking it into the update process.

Is there anything else that I can do to update my device?

### What device are you using? 
OGA 1.1

### What version of ArkOS are you on?
12082023

christianhaitian, 2024-03-16:
You’re supposed to scroll down to the bottom of the keyboard then press on the OK after typing ok.

[image]

wouterVE, 2024-03-16:
OK it seems on OGA the OK & Cancel buttons are not visible on the screen (probably due to other screen resolution).
See my screenshot below. When I press down I can press OK however so my problem is solved by this. 
[image]

So the solution is to press down on the ghost OK button ore maybe change the resolution for OGA?

Thanks for your help

christianhaitian, 2024-03-16:
It should be resolved after you complete this update.

closed 2024-03-16 by christianhaitian

wouterVE, 2024-03-17:
Indeed, now I can see the OK and Cancel buttons in the keyboard screen.

Thanks for your support