christianhaitian/arkos #995: Feature: Please, add picotron support

RobertoFreireFerrazPassos, 2024-03-17:
https://www.lexaloffle.com/picotron.php

RobertoFreireFerrazPassos, 2024-03-17:
https://www.lexaloffle.com/picotron.php?page=roadmap

Picotron 0.2 (2024 Q4)
    Raspi 64-bit
    Binary Exporters   // Windows, Mac, Linux, Raspi 64-bit
    Tooltray           // Run apps as widgets accessible from any workspace
    BBS cart switching // BBS carts can load & run other BBS carts

christianhaitian, 2024-03-17:
Where are the games?  and the Raspberry PI executable is not expected to be available until the last quarter of the year which would be some time between October and December.  Even if these deadlines are kept, there's no guarantee the Raspberry Pi binary will work on these devices if they end up using some specific raspberry pi features like broadcom specific libraries.

RobertoFreireFerrazPassos, 2024-03-17:
No games right now. Picotron has just released version 0.1 (2024-03-14)
Thanks for taking the time to answer so quickly

christianhaitian, 2024-03-17:
Well, let's see where this goes.  Since there's no games, it doesn't make sense to consider adding support for this until this has progressed more and there's actual games can be played with this.

closed 2024-03-17 by christianhaitian