christianhaitian/arkos #996: How do I download the firmware?

aksenav2, 2024-03-19:
Hello, how to download the firmware for the console r36s, to flash the flash drive?

christianhaitian, 2024-03-19:
Get the image from here: https://github.com/AeolusUX/ArkOS-R3XS
Follow the instructions here: https://ko-fi.com/post/Installation-Guide-for-ArkOS-v2-0-01272024-J3J6TVPH1

closed 2024-03-19 by christianhaitian

aksenav2, 2024-03-23:
After updating the ArkOs firmware, Retroarch changed interface, it is very inconvenient for me to use it. And that's how it became. 
[image]
How do I get it back to the previous one? Anyway, I've seen several themes on Amber Elec. 
[image]