christianhaitian/arkos #998: Virtual Keyboard

gateyex123, 2024-03-22:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

**Yes.**

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

**It is not. It's a feature request.**

5. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

**Yes, though this is not really an issue.**

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

**The feature is simply to add a hotkey to have a virtual keyboard overlay, globally if possible and a hotkey mapped to bring it up. This will be useful for places where a keyboard would be more fit to enter data or for a game. For example, with many PortMaster ports, a keyboard is needed, either for `sv_cheats` on Half-Life or GTA cheats.**

### How can the issue be reproduced? (If applicable)

N/A

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

N/A

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

**The latest. It says something-022024**

### Anything else you'd like to include that may help to review this issue or feature request?

Not really.
I tried installing matchbox-keyboard to see if it runs out of the box, but it does not. Other virtual keyboards require 50+MBs, like xvkbd, and 350+MBs, like florence, of dependencies and that by itself is a good indication they won't work and might ruin my system.

Are there maybe virtual keyboards tailored for minimal Linux distros like Ark or is this perhaps something that needs to be developed?
It's pretty niche, so I am assuming not.

gateyex123, 2024-03-22:
I am pretty clueless about linux for embedded systems, and all I currently know is that since there is no traditional display server, and the OS most probably just has direct access, then a custom solution would have to be made. Right?

christianhaitian, 2024-03-22:
Yes a custom solution of some sort would be needed for this.  Not something I have the skill for but perhaps someone will develop such a solution in the future.  Until then, programs will need to provide their own virtual keyboards.

closed 2024-03-22 by christianhaitian