christianhaitian/arkos #999: Build 351Files

aksenav2, 2024-03-22:
Hello, I want to compile 351Files, but I'm getting a mistake, what am I doing wrong? 
Still need a toolchain? Where can I download it? 
[image]

aksenav2, 2024-03-22:
[image]

christianhaitian, 2024-03-22:
You can build it on the device (https://github.com/christianhaitian/arkos/wiki/Building-packages-and-modules-on-your-device) or via a chroot (https://github.com/christianhaitian/arkos/wiki/Building#to-create-debian-based-chroots-in-a-linux-environment).  The RG351MP build should give a 640x480 resolution.

christianhaitian, 2024-03-22:
If you build on your device, I recommend using a separate SD card for that as the process for converting ArkOS into a build environment is destructive for your current setup.

closed 2024-03-22 by christianhaitian