christianhaitian/arkos #1000: Appearance and design of the retroarch emulator.

aksenav2, 2024-03-23:
After updating the ArkOs version,  changed the appearance of the retroarch emulator and became like this.
[image]
How do I get back to the old way back like this? 
[image]
On emberelec flashing, this is done very simply, and there are several different designs. Is there the same on ArkOs?

christianhaitian, 2024-03-23:
Change the menu driver. https://www.youtube.com/watch?v=zwDDj2I9OHI

closed 2024-03-23 by christianhaitian

aksenav2, 2024-03-23:
Thank you so much for your help, it works.