christianhaitian/arkos #1001: Wrong carousel position after ES settings change

dani7959, 2024-03-23:
Change a setting that requires ES to restart, such as: Main Menu > UI Settings > On-screen Help.
After restart the Carousel will be moved to a different position.
Expected behavior: Carousel should remain in the same position after settings change.

christianhaitian, 2024-03-23:
What unit and which theme is this occurring in?

dani7959, 2024-03-24:
It's a RG353VS.
This happens on the default Nes Box theme, as well as on the Replica theme.
Didn't test other themes but it doesn't look like it's theme-related. Most likely an ES bug.

christianhaitian, 2024-03-24:
Did you make sure to check the setting for the theme configuration is still set to 4:3?

christianhaitian, 2024-03-24:
When you say the carousel, you mean the system you were on before ES refreshes?  Like if you have Mame highlighted, then make these changes, screen refreshes, then it moves to another system?  If so, I'm not getting that on my end.

christianhaitian, 2024-03-24:
I've tried with the nes-box theme and the switch theme and I'm not getting this on my end.  Use ThemeMaster and update the theme to be sure it's not related to something there.

dani7959, 2024-03-25:
> When you say the carousel, you mean the system you were on before ES refreshes? Like if you have Mame highlighted, then make these changes, screen refreshes, then it moves to another system? If so, I'm not getting that on my end.

Yes, that is what happens.

Its's not happening when you manually restart ES. Only when you change a setting that automatically restarts ES (such as toggling the help system).

If you're not getting that, it might be something specific to my device.

closed 2024-03-25 by christianhaitian