christianhaitian/arkos #1003: Saves after v2.0 (03/29/2024) Update

czapa86, 2024-03-29:
Hello,

After today's update v2.0 (03/29/2024) on Rg353M device all saves and saves states are saved in core specific folder. Previously all saves were located in the root directory of rom file. This is causing old saves not to work. How to change it? I know the solution is to copy existing saves to that core folder but it will be time consuming.
Example: old saves are in /Roms/psx folder where now roms/psx/pcsx_rearmed folder

christianhaitian, 2024-03-29:
I'll review and address this asap today.  Must have been a default change by libretro with the new version.

czapa86, 2024-03-29:
Thanks! Just to add all systems/cores are afected, looks like retroarch update caused this.

christianhaitian, 2024-03-30:
This has been addressed with today's update.

closed 2024-03-30 by christianhaitian