christianhaitian/arkos #1005: Potentially update version numbering to YYYYMMDD format

fexofenadine, 2024-03-30:
Currently the version numbering is in a difficult to read format, MMDDYYYY, which doesn't sort, or read easily for non-US readers

If it could be changed in future to YYYYMMDD, or even YYYY-MM-DD, it would be both sortable and readable

Happy for you to ignore this as it's of little consequence, and the current naming/numbering might have been a conscious choice

Thank you!

christianhaitian, 2024-03-30:
Why would this need to be sorted?

fexofenadine, 2024-03-30:
[image]
like this for example

like i said it's no big deal its just that when the month comes first, all the versions are out of chronological order

christianhaitian, 2024-03-30:
Ahh I see.  Yea not something I will accommodate at this time but I understand the thought now.

closed 2024-03-30 by christianhaitian