christianhaitian/arkos #1006: Language

DiLZayd, 2024-03-30:
please add Ukranian? I can translate ark os

christianhaitian, 2024-03-30:
The following file can be used for this.
https://github.com/christianhaitian/EmulationStation-fcamod/blob/351v/resources/locale/es/emulationstation2.po

christianhaitian, 2024-04-02:
Hi,

Please use the emulationstation2.po file that's in the attachment.  It's more up to date.  What you can do is input the translations into the msgstr "" line for each of the msgid strings shown and save that.
[emulationstation2.zip](https://github.com/christianhaitian/arkos/files/14839436/emulationstation2.zip)

lauren7ino, 2024-04-02:
OK, here’s my best.

Didn’t test it, but I expect no translation problems.

￼

Thanks

--Laurentino Martins

lauren7ino, 2024-04-02:
I've posted the Zip on the other thread: 
https://github.com/christianhaitian/arkos/issues/1011

christianhaitian, 2024-06-15:
Any update on this?

closed 2024-06-20 by christianhaitian

jmas, 2025-11-16:
@christianhaitian Lexer error:
```
 % gpt-po --po ./emulationstation2.po
file:///Users/alex/.nvm/versions/node/v24.10.0/lib/node_modules/gpt-po/node_modules/gettext-parser/lib/poparser.js:221
            throw err;
            ^

SyntaxError: Error parsing PO data: Invalid key name ")" at line 404. This can be caused by an unescaped quote character in a msgid or msgstr value.
```

Also issue in line:
```
msgid "THE \"
```

Looks like *.PO file syntax broken. Could you check it.