christianhaitian/arkos #1007: Scraped box art does not persist after shutdown / reboot

hiruko-san, 2024-03-30:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes.

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Not related.

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
After I scraped box art for a given system (e.g. GameBoy, GameBoy Color, Playstation), it would be missing after a system shutdown / reboot.

I've checked in Theme> UI settings and set Gamelist view to Detailed, as well as in the individual games folder to Detailed.
I've enabled both Wifi & Remote Services.

I'm using Powkiddy RGB30 with 2 SD cards.

Even more frustrating is for Parasite Eve specifically (I happen to be playing that while testing), the savestates would be overwritten into blank states, and the memory card would need to be formatted. 

### How can the issue be reproduced? (If applicable)
Scraper> Screenscraper, Image Source: Box 2D, Box: none, Logo: none, Ratings: ON, Videos: On, Username&Password: is correct> Scrape Now> Only Missing Media> Start

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
03302024

### Anything else you'd like to include that may help to review this issue or feature request?
I have tried enabling & disabling Quick Mode but that didn't help.

christianhaitian, 2024-03-30:
This usually  happens for 1 of 2 reasons.

1. Improper shutdown of the device.   Are you just holding the power button to shutdown the device or hitting the reset button to reset and not shutting down via the ES start menu or hotkey or rebooting via the ES start menu?  If so, then the metadata is not getting saved before shutting down.  
2. You have a faulty sd card.  Are you using good SD cards?  Samsung and Sandisk cards are recommended for reliability.

Maybe the default pcsx-rearmed emulator has an issue with Parasite Eve.  You can try another emulator such as swanstation to see if that resolves the save issue.  If you need help on how to change the emulator, see the FAQ for the RGB30 in the wiki.

hiruko-san, 2024-03-30:
I've tried holding R3 + holding Power button (the hotkey to shutdown) device; holding R3 + short pressing the power button to put it to sleep mode; exiting the emulator and then going into the ES menu> Quit> shutdown.

I've switched both of my cards to SanDisk cards, 1st slot SD card (for the OS) is 32 GB; while the 2nd slot is 256 GB.

I tested it again & scraped one game's metadata by pressing Select while I highlighted the game> edit metadata> scrape> download image> save. 

I then go into the ES menu> Quit> shutdown.

After I boot back into ES, the metadata is gone, again.

I just can't seem to get it to persist.

hiruko-san, 2024-03-30:
I think I might have found the culprit. I've got a fake / bad quality sandisk card.
It's supposed to be a 256GB card and yet it shows up as 268GB total capacity.

I think the disk has been overwriting data in the background because it's actual size must be much smaller.

I'll test it again with a genuine card and will report back.

closed 2024-04-02 by hiruko-san