christianhaitian/arkos #1008: No video for STANDALONE-RICE on R36S

catallo, 2024-03-30:
On R36S with latest release of ArkOS (03302024): 

If 'STANDALONE-RICE' for N64 emulation is selected the game will start (sound works) but there is no video (black screen).

After some digging I found out that video works if the line

`ResolutionWidth = 480`

in the [Video-Rice] section of /home/ark/.config/mupen64plus/mupen64plus.cfg is removed.

christianhaitian, 2024-03-30:
Interesting as that line works fine with the rg351mp.  I don't support that device.  Copying @AeolusUX so he's aware and can accomodate that if needed in his custom ArkOS image.

AeolusUX, 2024-03-30:
> Interesting as that line works fine with the rg351mp. I don't support that device. Copying @AeolusUX so he's aware and can accomodate that if needed in his custom ArkOS image.

Will check when I start doing the update on the community image. It used to work before with that line.

closed 2024-03-30 by christianhaitian