christianhaitian/arkos #1009: Apple 2 emulators can't be launced from ES. Applewin core is missing a dependancy

zomgugoff, 2024-03-30:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
4. Yes

5. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Apple II emulators are not able to start. The 'apple2' section of es_systems.cfg is missing emulator and core entries. Also, applewin in Retroarch is failing to start. Logs show "[ERROR] Error(s): libslirp.so.0: cannot open shared object file: No such file or directory".

### How can the issue be reproduced? (If applicable)
Try to start an apple2 game, or try to start the applewin core in Retroarch

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 03302024

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-03-30:
I can't replicate this issue on my end.  All my testing shows the entries are added fine into ES and the resources needed by the emulators are included in the `/opt/applewin` folder and when launched with the `/usr/local/bin/apple2.sh` script, they launch without issue.  Do you have a custom es_systems.cfg you overwrote the previous updated es_systems.cfg with?

christianhaitian, 2024-03-30:
This is the entry that should have been included in your es_systems.cfg file from the 03292024 update:
```
	<system>
		<name>apple2</name>
		<fullname>Apple II</fullname>
		<path>/roms/apple2/</path>
		<extension>.dsk .DSK .sh .SH .do .DO .po .PO .apple2 .APPLE2 .zip .ZIP</extension>
		<command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/apple2.sh %EMULATOR% %CORE% %ROM%; sudo perfnorm</command>
		<emulators>
		   <emulator name="linapple">
		   </emulator>
		   <emulator name="retroarch">
		     <cores>
	           <core>applewin</core>	
	         </cores>
		   </emulator>
		   <emulator name="applewin">
		     <cores>
	           <core>scanlines</core>	
	           <core>no-scanlines</core>	
	         </cores>
		   </emulator>
	    </emulators>
	    <platform>apple2</platform>
	    <theme>apple2</theme>
	</system>
```
Also, the applewin2 core can't be started directly from within retroarch as it is still being worked on by the creator and requires various tricks to start that the apple2.sh file accounts for.

closed 2024-03-30 by christianhaitian