christianhaitian/arkos #1011: Translations

lauren7ino, 2024-04-02:
Hi Christian,

I've been using my RG353M in Portuguese (pt-PT) and I notice that some stuff is not translated. In main menu Options, All Games, Favorites, Last Played, (etc) are not translatable?
If not possible, where can I rename these to Portuguese?
I mean, are they in some text file right?

By the way the file pt\emulationstation2.po seems to have a syntax error in line 2143.

Thanks

christianhaitian, 2024-04-02:
Hi Lauren,

It's time to clean that up as there's been many changes to ES that's causing the translation issues you're seeing.  I was able to create a new updated template that can be used for these translations.  It's attached in the zip file to this message.  What you can do is input the translations into the `msgstr ""` line for each of the `msgid` strings shown and save that.  If you'd like to test this on your installation, you just need to SSH into your device and copy this `emulationstation2.po` file into the `/usr/bin/emulationstation/resources/locale/pt` folder, then restart emulationstation and see if the translations are complete.  Let me know your results because there could be some fields still missing or code needing to be adjusted in ES to support translation.

Thanks,
[emulationstation2.zip](https://github.com/christianhaitian/arkos/files/14839430/emulationstation2.zip)

lauren7ino, 2024-04-02:
OK, here’s my best.
Didn’t test it, but I expect no translation problems.
[emulationstation2.po.zip](https://github.com/christianhaitian/arkos/files/14841181/emulationstation2.po.zip)

christianhaitian, 2024-04-02:
Hi Lauren,

Sorry, the last one I posted wasn't complete enough.  Here's the correct one instead.

It's time to clean that up as there's been many changes to ES that's causing the translation issues you're seeing. I was able to create a new updated template that can be used for these translations. It's attached in the zip file to this message. What you can do is input the translations into the msgstr "" line for each of the msgid strings shown and save that. If you'd like to test this on your installation, you just need to SSH into your device and copy this emulationstation2.po file into the /usr/bin/emulationstation/resources/locale/pt folder, then restart emulationstation and see if the translations are complete. Let me know your results because there could be some fields still missing or code needing to be adjusted in ES to support translation.

Thanks,
[emulationstation2.zip](https://github.com/christianhaitian/arkos/files/14841765/emulationstation2.zip)

lauren7ino, 2024-04-02:
[(latest) emulationstation2.po.zip](https://github.com/christianhaitian/arkos/files/14843943/latest.emulationstation2.po.zip)

Here it goes. A professional translation (*)
Thanks!

(*) Actually part of my job

lauren7ino, 2024-04-02:
BTW, I use ForkLift on the Mac for accessing device files. Much user friendly.

lauren7ino, 2024-04-03:
Hi
Could not test the file in /usr/bin/emulationstation/resources/locale/pt
Always reports Permission Denied

lauren7ino, 2024-04-04:
OK, I managed to replace the file. Had to open the path with sudo.
I'm collection untranslated messages and doing some fixes.

lauren7ino, 2024-04-04:
Main screen:
All Games / Favorites / Last Played / etc.

Missing:
Panel Version: 2

On/Off icons?
Theme aspect ratio / color / scroll sound / on / off
UI Mode: FULL/KIOSK/etc
TRANSITION STYLE: Instant/Auto/etc.
Governor: Performance/ondemand/etc.
Log levels Default/Disabled/etc.

lauren7ino, 2024-04-10:
Hi
When will be the next release of ArkOS?
Are my translations good enough?
Thanks

christianhaitian, 2024-05-03:
> OK, I'm going to submit the new file tonight, but after that you make a batch search replace for these 2 words: OCCURED to OCCURRED CUSTOMISATION to CUSTOMIZATION (Mac highlighted these, otherwise I wouldn't have noticed) 😊

The next update will have these corrections in place.  You may need to update your translation file to account for that.

christianhaitian, 2024-05-03:
> Missing:
> 
> Main screen: All Games / Favorites / Last Played / etc.
> 
> Panel Version: 2
> 
> On/Off icons? Theme aspect ratio / color / scroll sound / on / off UI Mode: FULL/KIOSK/etc TRANSITION STYLE: Instant/Auto/etc. Governor: Performance/ondemand/etc. Log levels Default/Disabled/etc.

Not sure how best I will handle these.  Some like the Governors will not be able to be translated as it would affect the passing of the arguments to adjust performance.

lauren7ino, 2024-05-03:
HiI’m a bit offline in the next few days as I’m moving home and I don’t have anything in place.Thanks —Laurentino MartinsEscrito no telemóvel, peço desculpa por algum erro.

christianhaitian, 2024-05-04:
I did that.  It will reflect in the next update.

lauren7ino, 2024-05-27:
Hi Christian,
Q1: If I find any problems in the PT translation, can I edit and push the changes?
Q2: What is the full path of the PT translations?
Thanks

alhumbra, 2024-05-31:
Hi Christian,
I sent the Japanese translation to [EmulationStation-fcamod](https://github.com/christianhaitian/EmulationStation-fcamod/pull/9)
Is there a problem with this pullrequest?

christianhaitian, 2024-05-31:
> Hi Christian, I sent the Japanese translation to [EmulationStation-fcamod](https://github.com/christianhaitian/EmulationStation-fcamod/pull/9) Is there a problem with this pullrequest?

Best to recreate that using this instead.  https://github.com/christianhaitian/arkos/issues/1011#issuecomment-2032672947

It's more complete that the last file I provided for a template

alhumbra, 2024-05-31:
Hi Christian,

The pullrequest template is [#1001(comment)](https://github.com/christianhaitian/arkos/issues/1011#issuecomment-2032672947)
I submitted a translation on this basis.

I just checked with DIFF to be sure, but the only differences are in the translated text and Japanese settings, and I think the translation is working fine.

Thanks

christianhaitian, 2024-06-16:
> Hi Christian,
>
> The pullrequest template is [#1001(comment)](https://github.com/christianhaitian/arkos/issues/1011#issuecomment-2032672947) I submitted a translation on this basis.
>
> I just checked [...]

Ok.  I've merged it and will include it in the next update.

christianhaitian, 2024-06-16:
> Hi Christian, Q1: If I find any problems in the PT translation, can I edit and push the changes? Q2: What is the full path of the PT translations? Thanks

How would you intend to edit and push the change?
full path to it in the os is `/usr/bin/emulationstation/resources/locale/pt/emulationstation2.po`

lauren7ino, 2024-06-16:
I thinking in submitting the edit using the Github interface on the web.

[image]

christianhaitian, 2024-07-07:
@lauren7ino Is your translation completed yet?

lauren7ino, 2024-07-07:
Hi Christian,
Me and my wife caught Covid and has been a rough couple weeks.
I may be able to look into it tonight, but no promises.
Thanks

christianhaitian, 2024-07-07:
Wow.  Ok no rush.  Take care and get well to you both.

closed 2024-07-08 by christianhaitian