christianhaitian/arkos #1014: RG353VS initial boot faiulure

paladinperry, 2024-04-03:
RG353VS - Black - 16gb sandisk / 64gb kioxia

Failure to initial boot.
arkos will hang on "arkos"
arkos will get to the .... and hang
arkos will get to ...umount: /roms failure to mount

I've tried multiple reflashes , I have got the 1.4ghz quick fix but I do not have a readable partition to place it in
as inital boot fails to boot

christianhaitian, 2024-04-03:
Readable partition should be the BOOT fat32 partition.  If you're not at least seeing that on your computer, something is either wrong with the flashing process or the sd card itself.

paladinperry, 2024-04-03:
Cheers Christian. 

I'd agree but I've tried 4 different branded cards 2x new now.

I may not be seeing the boot partition because I have so many drive letters in use maybe ?

I don't know. I just know I can't get past those screens or that error message.

Is there a recommended older version to try for the RG353VS ?

paladinperry, 2024-04-04:
I only use ArkOS mate.

I'll send it back. I've only had it a few days. Is this a common issue with the RG353VS ?
I want this form factor and chipset so I can play Saturn and DC with ArkOS with it.
I had the R36S before, but the screen and buttons were awful. I had to get rid.

Thanks again mate

closed 2024-04-18 by christianhaitian