christianhaitian/arkos #1015: [Feature Request] Emulating Symbian\N-Gage games

elmoiv, 2024-04-04:
This emulator [EKA2L1](https://github.com/EKA2L1/EKA2L1) can run Nokia Symbian apps on Linux. This would add tons of N-Gage games if compatible.

I don't have that background on Linux but I think it may be possible as ArkOS is a Linux fork. The buttons may be mapped with optional on screen keyboard.

What is your thoughts?

christianhaitian, 2024-04-05:
Already looking at this and the this emulator requires qt for the ui to build which is not available in ArkOS.  There's an older version of this that can be built without QT that the developer has said I can try but is missing many recent improvements.  So I plan to try it and if it can run some N-Gage games decently, I may include it.

elmoiv, 2024-04-05:
Thank you very much.

u2020bullet, 2024-05-16:
Can't wait for this if it comes to fruition, used to have an N-Gage.

FrostenFire, 2025-02-14:
It would be great! I will keep an eye on the updates!