christianhaitian/arkos #1020: Shadow on RG353M

Wds777, 2024-04-07:
I instal arkos on RG353M following the step, using recommended micro sd card (im using sandisk ultra 16gb) few minutes after installed, screen get shadowing
[image]

christianhaitian, 2024-04-07:
Do you have a v1 screen?  You can check in the start menu under display settings and info, Panel Version.  If it says 1, then you have a v1 screen.  Sometimes those have issues with the dual screen version dtb that’s available in ArkOS.  What you can do is go to the  fat32 `BOOT` partition of your sd card, and rename the existing `rk3566-OC.dtb` file to `rk3566-OC.dtb.bak` the rename the `rk3566-OC.dtb.bright` to `rk3566-OC.dtb`.  Then boot up ArkOS and you shouldn’t have the issue anymore.

The fat32 boot partition on the sd card can be accessed directly from a mac, windows, or linux computer with a sd card reader/writer.  You can also access it via ssh to make this change.  Id you do access it via ssh, you’ll need to use sudo to make these changes.

Wds777, 2024-04-07:
[image]

Wds777, 2024-04-07:
Yes, panel version : 1. Lemme follow your guide and i'll get back to you for the report

Wds777, 2024-04-07:
Problem solved. Many thanks dev

closed 2024-04-07 by Wds777