christianhaitian/arkos #1021: Feature request: basic input test

CavemanJoe, 2024-04-07:
I have an RG353V whose joysticks might be getting worn out, or there might be some weirdness going on in software.  It'd be very nice if there were a very basic low-level input test screen that could tell me numeric X/Y values for the joysticks, so I could find dead spots in the pots and eliminate layers of software from my list of things to check.

I'm thinking something like you'd see on the switch test features of arcade and pinball games.  Could be useful to anyone trying to fix a partially-working handheld.

christianhaitian, 2024-04-07:
Would something like this work for your use case? <https://www.youtube.com/watch?v=phKoKU2lYFs>
If looking for a graphical solution, not sure one is available for this platform on linux just yet.

CavemanJoe, 2024-04-07:
Aye that'd do the job just fine.  Long term it'd be nice to have a graph of the joystick x/y so you could slowly move it and watch for dead spots, but this'd do for now no problem.

christianhaitian, 2024-04-18:
I've put together a solution for this.  If you could help me test this and provide feedback, it would be helpful.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot the device.

4. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then select Controller Tester., hit A.

You can then test your inputs.  You can test the rumble motor by pressing L3+R3 together.  Select+Start to exit the tool.

Let me know your results.

Thank you,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/15025661/arkosbackup.tar.gz)

closed 2024-04-22 by CavemanJoe