christianhaitian/arkos #1022: Feature request: Alternative button layouts

lauren7ino, 2024-04-08:
Hi Christian,

First of all, thank you for pointing me to the right direction in this post:
https://github.com/christianhaitian/arkos/issues/905#issuecomment-1912359369
I'm a happy gamer now :)

Request:

Please consider distributing an alternate DTB file with Xbox button layout.
A <=> B swapped
X <=> Y swapped
I know physical buttons can not be swapped (can they?) but I believe Xbox gamers would appreciate this.

Thank you

christianhaitian, 2024-04-08:
Good.  This is not something I will accommodate at this time as it would increase maintenance across 10+ devices that I don't want to manage unless this can be accomplished with simple text file changes which I wouldn't know how to accomplish.

closed 2024-04-08 by christianhaitian

lauren7ino, 2024-04-08:
Linux seems to have Kernel support for Device Tree Overlays
https://www.96boards.org/documentation/consumer/dragonboard/dragonboard410c/guides/dt-overlays.md.html

I would imagine it would be possible to have a base TDB with no A/B/X/Y buttons, a TDB for Nintendo layout (default) and TDB for Xbox, which could be swapped using a script in ArkOS.

lauren7ino, 2024-04-08:
Another ideia:
A script in ArkOS that swaps buttons between Nintendo and Xbox.
Could be acomplished by expanding the existing DTB:
  dtc -I dtb -O dts rk3566-OC.dtb -o rk3566-OC.dts
patches the generated DTS file to swap the buttons, then recompiles the DTB
  dtc -O dtb -o rk3566-OC.dtb -b -0 rk3566-OC.dts
reboots.
All indevice :)

lauren7ino, 2024-04-09:
Sorry, I don't know anything about RK3326 devices 🙁

Anyway, I ended up just swapping A/B, because X/Y don't bother anyone.
That's what most UIs do anyway, EmulationStation, PortMaster and RetroArch only have A/B swap settings.