christianhaitian/arkos #1025: Emmc doubts

lauren7ino, 2024-04-11:
Hi Christian,

Regarding transferring ArkOS to the 353M EMMC as described here
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG353M#q-can-i-load-arkos-onto-the-internal-emmc-storage

Q1: After this operation is complete and working, how do you access the directories in the EMMC in case ArkOS for some reason doesn't boot?
Q2: Any other limitations I should be aware of?

Thank you

christianhaitian, 2024-04-11:
Login as usual and make your changes.  It will look like tf1 except root will be linked to /dev/mmcblk0 instead of /dev/mmcblk1
No limitation from my point of view if you don't like Android on this device.  If you do link Android on this device, then you can't have Android and Linux at the same time with this setup.

closed 2024-04-11 by christianhaitian

lauren7ino, 2024-04-12:
Thanks I was not aware of this.
I see two differences: 
1. We can no longer popup the SD1 Card and pop it in Linux for more complex editing/compiling.
2. If ArkOS gets messed up, or the boot partition files, it's unrecoverable (?)

Advantages:
1. Stability (probably?) since EMMC is much more reliable that SD Cards.
2. Launch speed.

Anyway, you may like to know I've tested the process described in your Wiki and worked perfectly.
I'm now an EMMC user :)
Thank you!

christianhaitian, 2024-04-12:
You can recover the emmc.  Just boot from arkos in tf1 as it still takes precedence on boot if a sd card is inserted.

lauren7ino, 2024-04-12:
> You can recover the emmc. Just boot from arkos in tf1 as it still takes precedence on boot if a sd card is inserted.

Yes, but from that SD card can I access files in the emmc?

christianhaitian, 2024-04-12:
Yes

christianhaitian, 2024-04-12:
just mount `/dev/mmcblk0p4`