christianhaitian/arkos #1026: RGB30 sd card 2

widiot1, 2024-04-11:
I recently bought a RGB30(new model with colored buttons) and cant get sd card 2 to be recognised. Ive tried half a dozen different brands and cards ranging from cheap chinese no name to sandisk ultra/PNY. None of them get detected by my RGB30. I decided to hook it up to a terminal and dmesg brings up the following

[  247.689574] mmc_host mmc2: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
[  247.722759] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 300000Hz
[  247.722794] mmc_host mmc2: Bus speed (slot 0) = 375000Hz (slot req 300000Hz, actual 187500HZ div = 1)
[  247.733556] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 300000Hz
[  247.736862] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 300000Hz
[  247.740269] rockchip-dmc dmc: cpu3 policy NULL
[  247.760237] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 200000Hz
[  247.760266] mmc_host mmc2: Bus speed (slot 0) = 375000Hz (slot req 200000Hz, actual 187500HZ div = 1)
[  247.772892] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 200000Hz
[  247.776318] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 200000Hz
[  247.793680] rockchip-dmc dmc: cpu0 policy NULL
[  247.800021] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 100000Hz
[  247.800047] mmc_host mmc2: Bus speed (slot 0) = 375000Hz (slot req 100000Hz, actual 93750HZ div = 2)
[  247.812679] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 100000Hz
[  247.816774] dwmmc_rockchip fe2c0000.dwmmc: failed to set rate 100000Hz

Im not sure if this is an issue with my device(hardware fault) or an issue with the rockchip kernel. As I seem to be the only person with this issue Im guessing I just got a faulty unit but just wanted confirmation before I look at returning it

christianhaitian, 2024-04-11:
You didn't state if you did the following step: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rgb30#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games

If you did, then you may have a faulty slot.  Does the 2nd slot work with the stock os and a second sd card inserted?

widiot1, 2024-04-12:
I've tried it with multiple OS's now and can't get the 2nd slot to show up in any of them.  All cards have been formatted and reset many times and have tried every filesystem I can. Looks like my card slot is faulty. Thanks for the reply and thanks for the great OS.

christianhaitian, 2024-04-12:
Sorry to hear that.  Good luck with replacing the device.  If you don’t replace it, it will work fine with a single sd card setup as well.  No difference in capabilities except if you need to reflash the os, you would need to copy your roms and saves back to the card

closed 2024-04-12 by christianhaitian

widiot1, 2024-04-28:
Just wanted to add a follow up to this. 
I decided to keep my RGB30 and just run off one sd card. This works great with arkos , and I've been playing with rocknix too. 
I left an SD card in the second slot and it's still unseen by arkos and rocknix. However in rocknix there is an option to "remove SD card" so out of curiosity I selected that. Then when I refresh game list it miraculously finds all the roms on the second SD card. It's bizarre and makes no sense but it proves that the SD card slot is physically fine and the sd cards work. They just don't get detected automatically by any os and the only way I can use it is to use the "eject SD card" from the menu and then refresh the game list. It's definitely odd behaviour and seems unique to my device and I have zero idea why or what is going on. But anyway I'm fine using the single SD card as a solution and just thought I'd share my discovery