christianhaitian/arkos #1027: Online updater error: You didn't type OK

sid8tive, 2024-04-11:
### Describe the Issue (If applicable)
When attempting to update ArkOS using the online updater, it throws the following error after typing 'OK' (or 'ok') in the confirmation dialog screen: "You didn't type OK. This script will exit now and no changes have been made from this process."

There was a second SD card mounted, so I even ran the 'switch to main SD for ROMS' prior to running the updater. Now only one SD card is being used/inserted, yet the above error.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

R35S

### What version of ArkOS are you on?
v2.0 (02232024)

christianhaitian, 2024-04-11:
I can't replicate this issue on my end.  Perhaps try a reflash and update from that?

lauren7ino, 2024-04-12:
Something like this?
https://www.reddit.com/r/RG353V/comments/18h70jg/unable_to_update_arkos/

Can you see the popup keyboard?

sid8tive, 2024-04-12:
Update: So, before reflashing the microSD, I thought of trying the Updater one last time. This time it worked!

(My personal suspicion: the CDN may have been unavailable for whatever reason, but the script kept pointing to me not typing OK for some reason instead. That explains why it's working when trying at another time like I did just now.)

Closing this thread since issue is now resolved.

closed 2024-04-12 by sid8tive