christianhaitian/arkos #1028: Arkos browser - Can't follow instructions

lauren7ino, 2024-04-12:
I Christian,

Yesterday I tried to follow these instructions with no success:
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---CHI#q-how-do-i-use-arkos-browser-for-managing-my-roms-via-a-web-browser

I'm missing something here.

I'm specially confused with http://chi/
My browser keeps trying to open http://chi.com
???

christianhaitian, 2024-04-12:
You have a Chi?

lauren7ino, 2024-04-12:
Lol. I had no what idea what was a 'Chi'.
I have to try later again when I get back home.
But I did try a http://<ip> (and https) with no success either.

lauren7ino, 2024-04-12:
Got it working after reconecting the wifi and reenabling the remote services.

Maybe in the Wiki you could fix that Chi.
"Once enabled, go to your desktop or mobile browser and type http://CHI"
(* Replace CHI by your device name)

closed 2024-04-12 by lauren7ino

christianhaitian, 2024-04-13:
Every supported unit has it's own FAQ with their own default system name.

lauren7ino, 2024-04-13:
That explains it!
I didn't realize I was not looking at the 353m faq.
Sorry, my fault.