christianhaitian/arkos #1029: Help in device

lauren7ino, 2024-04-12:
Hi Christian,
Do you think it would be possible to create a 'Help' entry in ES Carousel for help regarding hotkeys on ArkOS?
I know the information is available in your Wiki and other locations, but it's tedious to find the information, specially while commuting standing up juggling the phone an the console.
Information like global hotkeys, plus troubleshooting, like the combo to reset audio settings, accessing BaRT, etc.
Just an ideia.
Thanks

christianhaitian, 2024-04-12:
Something I've toyed with but not something I will focus on anytime soon.  With my daily driver, I just play my games or watch movies which requires very few hotkeys.  If you're out and about needing information on all of this, you're not playing games, you're tinkering which should probably be done when you're in a setting that allows you to do all this.  Not while commuting and standing up.

closed 2024-04-12 by christianhaitian

lauren7ino, 2024-04-13:
The audio problem happens randomly, but seems to be somehow connected to Bluetooth headphones.

lauren7ino, 2024-04-13:
OK can I make a new suggestion instead?

Create a script in the Options menu to fully reset the audio, using all the tricks you described in this post, including replacing the config files:

https://github.com/christianhaitian/arkos/issues/987#issuecomment-1995441276

Thanks! Sorry the basic English

christianhaitian, 2024-04-13:
That’s a great idea.  I’ll inlude this in the next update

lauren7ino, 2024-04-13:
Make sure the script turns off Bluetooth. It may help too.

christianhaitian, 2024-04-13:
I won’t do that but i can provide a warning that it will reset audio to come out from the speakers and they will have to reconnect any headphones if they’re connected st the time the reset is initiated.

lauren7ino, 2024-04-25:
Hi
Just to let you know that Apple AirPods Pro (Bluetooth) seem to be reconnecting much better that the last time I tried months ago.
It’s true I frequently need to manually reactivate them, but at least I no longer need to repair them.
Thanks