christianhaitian/arkos #1031: Names platforms/emulator

aksenav2, 2024-04-12:
Hello, I liked the Amber theme and I adapted it for Arcos.  Please tell me where are located in the arkos system, the names of the platforms/emulators. I didn't find them in the themes resources. There are themes in which these names are spelled out, and there are themes where the names are taken from the system itself.
[image]

christianhaitian, 2024-04-12:
You can check `/etc/emulationstation/es_systems.cfg`.  The theme names are in the <theme></theme> xml tags for each system.

closed 2024-04-12 by christianhaitian

aksenav2, 2024-04-13:
> You can check `/etc/emulationstation/es_systems.cfg`. The theme names are in the xml tags for each system.

I've already searched in this file, there are names for everyone, except 'all games' 'last played' 'favorites'

aksenav2, 2024-04-13:
> See here for an example of correct theme names. https://github.com/Jetup13/es-theme-nes-box
> 
> it’s auto-allgames, auto-lastplayed, and auto-favorites respectively.

I know what they're called, but I want to translate them into another language, but I can't find where to do it. This is not in the resources of the themes, the names are taken from somewhere in the system.