christianhaitian/arkos #1032: Como eu posso criar um programa para rodar em sdl2?

BiancaRerre, 2024-04-13:
How can I create a program to run on sdl2? I wanted to somehow learn how to create programs to run on the portable console, but I can't even make a hello world
I've tried using python3, 
I was able to make a blue screen, with a gray retainer inside, 
However, I tried to write "hello world" inside and couldn't, because it gives an error, what can I do?

Here's the code for the blue background with gray rectangle:
[image]

`import sdl2
import sdl2.ext

# Configurações da tela
SCREEN_WIDTH = 480
SCREEN_HEIGHT = 320

# Inicialização do SDL
sdl2.ext.init()

# Criação da janela
window = sdl2.ext.Window("Desenho direto na tela", size=(SCREEN_WIDTH, SCREEN_HEIGHT))
window.show()

# Criação do renderizador
renderer = sdl2.ext.Renderer(window)

# Cor do quadrado (azul)
BLUE = sdl2.ext.Color(0, 0, 255)
# Cor do retângulo (cinza)
GRAY = sdl2.ext.Color(100, 100, 100)

# Desenha o quadrado azul e o retângulo cinza
def draw_shapes(renderer):
    renderer.clear()

    # Desenha o quadrado azul
    renderer.fill((0, 0, SCREEN_WIDTH, SCREEN_HEIGHT), BLUE)

    # Calcula as coordenadas para o retângulo cinza no centro da tela
    rect_width = 100
    rect_height = 50
    rect_x = (SCREEN_WIDTH - rect_width) // 2
    rect_y = (SCREEN_HEIGHT - rect_height) // 2

    # Desenha o retângulo cinza
    renderer.fill((rect_x, rect_y, rect_width, rect_height), GRAY)

    renderer.present()

# Loop principal
running = True
while running:
    for event in sdl2.ext.get_events():
        if event.type == sdl2.SDL_QUIT:
            running = False

    draw_shapes(renderer)

# Finalização do SDL
sdl2.ext.quit()
`

And here's the code I tried to write hello world:
`import sdl2
import sdl2.ext

# Screen settings
SCREEN_WIDTH = 480
SCREEN_HEIGHT = 320

# Initialize SDL
sdl2.ext.init()

# Create window
window = sdl2.ext.Window("Direct Drawing on Screen", size=(SCREEN_WIDTH, SCREEN_HEIGHT))
window.show()

# Create renderer
renderer = sdl2.ext.Renderer(window)

# Square color (blue)
BLUE = sdl2.ext.Color(0, 0, 255)
# Rectangle color (gray)
GRAY = sdl2.ext.Color(100, 100, 100)
# Text color (white)
WHITE = sdl2.ext.Color(255, 255, 255)

# Draw blue square, gray rectangle, and "Hello World" text
def draw_shapes(renderer):
    renderer.clear()

    # Draw blue square
    renderer.fill((0, 0, SCREEN_WIDTH, SCREEN_HEIGHT), BLUE)

    # Calculate coordinates for the gray rectangle in the center of the screen
    rect_width = 200
    rect_height = 100
    rect_x = (SCREEN_WIDTH - rect_width) // 2
    rect_y = (SCREEN_HEIGHT - rect_height) // 2

    # Draw gray rectangle
    renderer.fill((rect_x, rect_y, rect_width, rect_height), GRAY)

    # Render "Hello World" text in the center of the rectangle
    font_manager = sdl2.ext.FontManager(font_path=None, size=30)
    text_surface = font_manager.render("Hello World", color=WHITE)
    text_width = text_surface.size[0]
    text_height = text_surface.size[1]
    text_x = rect_x + (rect_width - text_width) // 2
    text_y = rect_y + (rect_height - text_height) // 2
    renderer.copy(text_surface, dstrect=(text_x, text_y, text_width, text_height))

    renderer.present()

# Main loop
running = True
while running:
    for event in sdl2.ext.get_events():
        if event.type == sdl2.SDL_QUIT:
            running = False

    draw_shapes(renderer)

# Quit SDL
sdl2.ext.quit()
`

I tried to use pygame, but I couldn't even install lib pygame because it gave an error when I try to install pygame...
Are there other ways or some easy solution for me to create programs that run on my handheld device, I wanted to be able to contribute to the community with some programs that can add some non-existent functionality, or even create a little game

christianhaitian, 2024-04-14:
I can only guide you to how to convert ArkOS to a development environment on your device by following the instructions here: https://github.com/christianhaitian/arkos/wiki/Building-packages-and-modules-on-your-device

I'm not versed on writing programs.  I just know how to compile them with some instruction.  You may have better luck talking with some of the folks in the PortMaster Discord.  The invite link to it is here: https://discord.gg/JxYBp9HTAY

Good luck.

closed 2024-04-14 by christianhaitian

lauren7ino, 2024-04-14:
Also taking a peek on existing graphic programs that run on ArkOS, like PortMaster, etc.