christianhaitian/arkos #1033: Strange Powkiddy RGB30 and ArkOS start up problem.

Mika73, 2024-04-15:
I have strange problem with RGB30 and ArkOS. 
Sometimes it doesn’t start up. I get either black screen or like this morning it stuck when loading Intellivision at start up.

After loading bar stuck to Intellivision I power off and on again. All I get was black screen. 

I know it sounds I have bad memorycard or corrupted memorycard. 

But if I remove memory cards and insert JelOS cards that come with it. It boot with those. After this if insert ArkOS cards back those work. 

It doesn’t help that eject ArkOS cards and reinsert those. I have to boot JelOS cards between and that fix problem. 

I don’t understand… it usually works everything ok but sometimes this happens. 

I have updated latest ArkOS version with WiFi update. 

### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

Mika73, 2024-04-15:
> What is the make and model of the sd cards you’re using?  If they’re are not genuine Sandisk or Samsung sd cards, try that first.  If they are and your still encountering this issue, then your device just might not be fully compatible with ArkOS. Good luck.

Kingston memory cards.

christianhaitian, 2024-04-15:
Kingston memory cards may not work well with ArkOS.  I've never used them with the OS so I don't know how well they will work or not.

christianhaitian, 2024-04-15:
Any Sandisk micro sd card should be fine as long as it is genuine.  I personally use the Ultra ones.

Mika73, 2024-04-15:
Other people seem to have same problem even with Sandisk cards. This is just what I am having. Don’t understand how boot Jelos between fix problem. But it works.