christianhaitian/arkos #1034: RA Start Core missing in many cores

lauren7ino, 2024-04-15:
Hi
I noticed that in RetroArch after manually loading a Core, many Cores don't display a Start Core main menu entry.
I'm wondering if there is something missing in the those cores configuration.

Example: Loading "Sinclair - ZX Spectrum (Fuse)" doesn't display a Start Core menu entry, but it should, because that's the only way to access the ZX Spectrum Basic.
Commodore 64 on the other hand works.
Amiga, it depends on the core loaded.

christianhaitian, 2024-04-15:
Then it's a limitation of the core.  Not something I can control.  Perhaps check with the core's creator for more assistance on this.

closed 2024-04-15 by christianhaitian

lauren7ino, 2024-04-16:
Thank you for your help.
I'm having difficulty finding the fuse support git page.

lauren7ino, 2024-04-16:
For future reference:
I've managed to boot FUSE Sinclair Spectrum without loading any game, by installing an empty game file (attached).

[image]
[!Boot.tzx.zip](https://github.com/christianhaitian/arkos/files/15003028/Boot.tzx.zip)