christianhaitian/arkos #1036: RGB30V2 cannot be shut down normally

marukochinn, 2024-04-17:
RGB30V2 cannot be shut down normally. When arkos is upgraded to the latest version 30032024, the shutdown option will change to restart. I hope there will be a correction provided, thank you!

christianhaitian, 2024-04-17:
I can’t replicate this issue on my end.  Did you recently try any other OS before this issue occurred?  If not, does this issue happen with any other OS?

marukochinn, 2024-04-17:
I was using rocknix OS before and this problem did not occur.

marukochinn, 2024-04-17:
After I reinstalled ArkOS_RGB30_v2.0_10162023, I was able to shut down normally... So don't upgrade yet?

marukochinn, 2024-04-17:
After updating 03302024, I started not being able to shut down again.

christianhaitian, 2024-04-17:
In the fat32 boot partition, Copy the .dtb file contained in the attached zip file into the boot partition overwriting the existing one there.  Then boot the device and see if it can shut down properly after that.  If it does, boot the device and see if it shutdown properly again or if the issue returns.  Let me know you results.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/15014588/rk3566-OC.zip)

marukochinn, 2024-04-17:
After replacing the rk3566-OC.dtb, I was finally able to shut down normally, and then tried several times to shut down normally. Thank you for providing the solution.

christianhaitian, 2024-04-17:
Odd.  The issue should have returned as the OS is supposed to check if it's a RGB30 V2 unit and replace that file with a v2 dtb for improved performance.  You may run into really slow performance with dreamcast and saturn.  For dreamcast, you can confirm with a game like power stone.  That should run between 50fps to 60 fps.  if it runs at half that speed, you're performance is capped

christianhaitian, 2024-04-17:
Oh correction.  It wouldn't.  it puts a check in place to prevent that dtb change a second time.  So the issue is because of the v2 dtb.  Good to know.  Thanks for helping me test this.  I'll have to see how best to resolve the shutdown and performance issue with v2 units.

closed 2024-04-17 by christianhaitian

marukochinn, 2024-04-17:
I tried DOA2 and power stone, both are as smooth as before.

christianhaitian, 2024-04-17:
🤯

marukochinn, 2024-04-17:
I just discovered that if you turn on the <switch power button tap to off>, pressing the switch will change it to restart again. Then if you use normal shutdown, it will also return to restart again.

marukochinn, 2024-04-18:
It's strange that I can't fix the problem again using the correction file you gave me, even if I reinstall the stock system.

christianhaitian, 2024-04-24:
I got a v2 unit in yesterday and the unit boots and shutsdown without issue on my end

marukochinn, 2024-04-24:
I discovered a very interesting thing. When I reinstalled the arkos system, I still couldn't shut it down normally. But if I use another micro sd card with a rocknix system to run it once, and then shut down and replace it with arkos. Then After that, the host can be turned on and off normally, even if it is set to shut down with one button press.

christianhaitian, 2024-04-26:
If you encounter this issue again, can you try the rk3566-OC.dtb file from here and let me know if it resolves the issue for you?
https://github.com/christianhaitian/arkos/issues/1046#issuecomment-2079183985

christianhaitian, 2024-04-26:
Have them try that then if possible