christianhaitian/arkos #1037: How to apply EmulationStation Command Line Argumements

canyoubarrett, 2024-04-18:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
yes
3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? no

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

i need to pass the no-splash argument at emulationstations startup

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

rg351v,

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

newest release

### Anything else you'd like to include that may help to review this issue or feature request?

thank you for taking time.

christianhaitian, 2024-04-18:
You have to include it in the `/usr/bin/emulationstation/emulationstation.sh` script.  If you'd like to test the capability first, you can do `sudo systemctl stop emulationstation` first then change into the `/usr/bin/emulationstation` directory, then do `./emulationstation <commandline>` to see what the command line done.

closed 2024-04-18 by christianhaitian

canyoubarrett, 2024-04-18:
im sorry i couldnt find the file emulationstation referenced in emulationstation.sh to add the argument

christianhaitian, 2024-04-18:
The line that has the following:
`"$esdir/emulationstation" "$@"`

canyoubarrett, 2024-04-18:
your patience knows no bounds. thank you