christianhaitian/arkos #1038: R36S - Keeps booting to fake-08 core with no game, can't exit, quick mode keeps trying to resume

rcampbel3, 2024-04-19:
I'm running

ArkOS 2.0 - 03302024 - on an R36S.
I've enabled quick mode. I've re-enabled hotkeys
Everything's been working quite well until yesterday. I ran a pico-8 game, it fell back to using the fake-08 core, which is fine... but then I was unable to exit the core with select+start. I did a hard power off, reboot, and ended back in fake-08 core with no game loaded. I looked online at all of the hotkeys and different ways to power off and have pretty much tried them all yet every time I boot, I end up resuming <NO GAME> in fake-08 core. Pressing the 'start' button brings up the fake-08 menu with these options:

continue
reset cart
exit to menu
exit to settings

Exit to settings has not option to quit.
Exit to menu shows me:

FAKE-08

WELCOME TO FAKE-08
...
place pb carts in ~/pbcarts
--no carts found

close window to exit

>

there's no way I can close this window, so I'm currently stuck in an infinite loop :( 

This seems like either a misconfiguration for respecting select+start for exit in this core (or possibly on my device), a bug with fake-08 core, or an enhancement request for quick mode - which I would hope can offer a failsafe boot normally option if a key is held during boot.

What I'd like to know is what file can I edit or delete to disable or bypass the quick mode and simply proceed with a normal boot. Or.. how I can disable quick mode from the terminal on a mounted microsd card. I haven't looked into the enable quick mode scripts yet - I feel like that will reveal the answer, but I have not found any posts online with a concise answer to "How do I stop ArkOS from booting to previous core/quick save when I can't exit the core?" other than people suggesting using a clean image... I'm a long time UNIX guy... that feels like admitting defeat and giving up, and this feels like a simple problem with a simple fix.

I have an image I generated with dd a few days ago, so I know I can recover, but I'd like to figure this out and prevent others from getting stuck.

rcampbel3, 2024-04-19:
I solved the immediate problem.

Mount OS card.
go to root/home/ark/.config directory
remove the file 'lastgame.sh'

Now... let's figure out how to prevent not being able to exit core or how to bypass quick mode on boot

christianhaitian, 2024-04-19:
You could have held the B button during boot and it would bring up the BaRT menu.  From there, you could have booted to Emulationstation which should have gotten you out of that state.

closed 2024-04-19 by christianhaitian

5310, 2024-12-05:
I faced this same issue, and had to [delete `lastgame.sh`](https://github.com/christianhaitian/arkos/issues/1038#issuecomment-2066997773) to be able to boot into a usable system as well. Thank you very much for the solution, OP! 

I must also need to mention, **while stuck in Fake08 Quick Mode boot-loop, I could _not_ hold the B button down to boot into BaRT**, [as instructed](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351MP#a-when-you-do-a-safe-shutdown-r3-button--power-button-while-in-a-retroarch-or-retroarch32-emulator-an-auto-savestate-will-be-created-of-the-current-game-and-the-unit-will-shutdown--upon-power-on-the-system-will-autoboot-into-the-last-game-and-resume-the-auto-savestate--to-enable-this-mode-select-enable-quick-mode-from-the-optionsadvanced-menu-section-in-emulationstation--reminder---this-only-works-with-retroarch-and-retroarch32-based-emulators--as-of-9212023-this-also-works-with-the-pico-8-standalone-emulator-as-well--you-can-boot-directly-into-the-last-pico-8-game-you-were-playing-or-into-splore--of-course-no-auto-savestate-will-be-available-since-the-emulator-doesnt-support-savestates). Nor could I hold down R1 to then wipe the save-state as otherwise instructed by ArkOS, (presumably because there is no save-state here?). Having gotten unstuck, BaRT now works as before. 

I've decided to disable Quick Mode, sadly.

joshsedl, 2024-12-31:
Thanks for the info, everybody! I just had a similar issue, where I was stuck in the default dreamcast core playing crazy taxi. Every time I was trying to simply close the game, an empty shell with a blinking blank appeared and I couldn't get back to emulation station.

> Nor could I hold down R1 to then wipe the save-state as otherwise instructed by ArkOS

Is there a section about this on the wiki? I couldn't find anything, but maybe I just didn't look thoroughly. Do you still have the link of this?
At least for me, I also enabled the option to shut down the device through tapping the power button. Maybe that is what caused it