christianhaitian/arkos #1040: Stick drift, software based

Zekul9, 2024-04-21:
R36S arkos 2.0 03302024
Having a very hard time nailing down the issue and being able to make it reproducible. Many users on forums have reported this issue of stick drift, even after replacing the sticks with hall effect the drift remains. In all cores and ports.
I saw the previous issue reported here asking for a input checker. I'm not sure I'm advanced enough to implement whatever solution that user was talking about. Something to calibrate the sticks could be vital and I'm surprised it hasn't come up yet.

christianhaitian, 2024-04-21:
I don't support the R36s units so I can not help much with your issue.  You can try here for more assistance if help is available there:  https://github.com/AeolusUX/ArkOS-R3XS  You're welcome to help test this input tester here: https://github.com/christianhaitian/arkos/issues/1021#issuecomment-2063885581
I have no solution for stick calibration.  You can try what this user has tried here for calibration.  https://github.com/christianhaitian/arkos/issues/1023#issuecomment-2046195241

closed 2024-04-21 by christianhaitian

Zekul9, 2024-04-28:
Ok. Pretend I have an rg351 then. No matter what I do I have drift. I never had this problem months ago. Now no matter regular sticks or hall effect when you press Right it does not return to center. Mostly only happens with right direction. Doesnt matter if I use a fresh install of arkos. I am using the same game sd though.
Comment 1023 is absolutely useless for me as its chock full of jargon.