christianhaitian/arkos #1043: [FEATURE REQUEST] Enable zram on devices with 1gb of ram

alekolo7, 2024-04-24:
Some newer postmaster games (minetest for sure, probably more to come) require more than 1gb of ram to fully work. 
A possible solution to that problem would be to enable (or make a setting to enable) zram.
By reserving 256mb of ram it would be possible to gain 1gb of ram disk (thanks to on-the-fly compression) and though to run some more memory-heavy titles without sacrificing so much performance as with normal swap memory, which involves offload to slow micro-SD card.
This change wouldn't be hard to implement as zram is embedded in most of modern Linux kernels and remaining 768mb of non-compressed ram would be enough to run retro emulators with no problem.

christianhaitian, 2024-04-24:
How do you do this?

alekolo7, 2024-04-25:
Here are some resources on that: 
https://linuxreviews.org/Zram https://www.kernel.org/doc/Documentation/blockdev/zram.txt 
https://www.reddit.com/r/lowendgaming/comments/13d5brx/enable_zram_on_low_end_linux_machines_for_a_free/

christianhaitian, 2024-04-25:
Have you tested this yet to see if this would be a benefit to PortMaster?  If so, What's you recommended process for making this option available?  If you haven't tested this, can you test this and provide the suggested configuration for this?

alekolo7, 2024-04-25:
Sure. I just set it up and noticed an mayor improvement in minetest. With normal swap I got 10-12 fps with many stutters, now with zram 15-25fps with less stutters. Without swap and zram it wouldn't even boot. 

As for enabling it, it's a matter of editing /usr/bin/init-zram-swapping. I attached my setup that works well for me ([init-zram-swapping.txt](https://github.com/christianhaitian/arkos/files/15122049/init-zram-swapping.txt)). 

There's one thing to be considered though. Zram generally makes effective ram larger, but it is archieved by compressing and then decompressing data, which is cpu-consuming. So if I reserved 512mb from my 1gb of ram for zram, I should have ~2gb effective total memory, but any usage that exceeds 512mb (the unreserved part) gets compressed and some performance is lost. So I think it should be implemented as a setting rather than being enabled by default. Emulators that use between 512mb and 1gb of ram may underperform a bit for example (haven't tested it, gonna do it tomorrow).

christianhaitian, 2024-04-25:
You think it should be something that's enabled and disabled on a per game basis?

alekolo7, 2024-04-25:
No, I don't think so. Enabling/disabling swap every time you launch/quit a game is too much I think. Also, I tested God of War on PPSSPP and there was no performance difference with zram enabled, so maybe enabling it system-wide (even by default) would be the best option? Requires some more testing though.

christianhaitian, 2024-04-25:
For emulation, the extra ram won't make a difference unless the unit has less than 1GB total available which none of the devices supported with ArkOS are concerned of.  Since it's best for certain limited ports only, I wonder if the portmaster group would consider it for implementation for certain ports that can benefit from it.

WinUser-Generic, 2024-06-04:
> For emulation, the extra ram won't make a difference unless the unit has less than 1GB total available which none of the devices supported with ArkOS are concerned of. Since it's best for certain [...]

and using for purpose to increase vram and continue getrting free memory dont gain any performance?

christianhaitian, 2024-06-04:
For the emulators supported, no.  None of them benefit from anymore than 1GB ram.  CPU speed is normally the limitation on performance.

closed 2024-06-07 by christianhaitian

CRCaue, 2025-07-15:
Sorry for intruding on this, I'm not even sure if anyone will be notified of this comment. But I'm currently trying to optimize PS1 games in ArkOS for R36S and I believe I have an interesting use case for ZRam.

I'm using CTR as test sample, running it after converting BIN/CUE image to CHD. I was able to squeeze performance from Retroarch with multiple settings (Even reducing the Audio Re-sampler quality helped), but maybe the main strat here is to Pre-Load the image file to memory.

CTR's CHD has 310.560 KB of size. Trying to preload it, I got an error in Retroarch. Maybe it is possible to improve RAM usage with ZRam, as other users reported improvements in low-end Linux builds (Not necessarily handhelds or game-focused machines). Preloading the image file should improve overall speed, as we know that CPU->RAM communication is faster than CPU->HDD/SDD.

I'm currently trying to activate ZRam in my R36S to check if this would help PS1 emulation in any way. I'll gladly report back if I get good results on it. I just don't think this should be discarded straightaway, as most Raspberry Pi Batocera builds use ZRam and get great results from it.

bmdhacks, 2025-09-02:
Actually, this is my /etc/default/zramswap that works well on my r36s.  The ALGO field doesn't do anything on the 4.4 kernel but if a newer kernel comes along this would be nice.

`
# Specifies a static amount of RAM that should be used for
# the ZRAM devices, this is in MiB
ALLOCATION=3072

# Compression algorithm selection
# speed: lz4 > zstd > lzo
# compression: zstd > lzo > lz4
# This is not inclusive of all that is available in latest kernels
ALGO=zstd
`

Luis-Kiss, 2025-09-29:
Init on Boot.