christianhaitian/arkos #1044: Request - Different icon for folders

lauren7ino, 2024-04-24:
Hi Christian,

Can you please change the prefix icon of folders in EmulationStation?

The current star icon for folders is exactly the same as a for favorite game.
In my case, I have the multi-disk Amiga games in each folder, like so (example game Gods)

/Amiga/Gods/Gods.m3u              <= Created by me so that UAE4ARM can swap disks
/Amiga/Gods/Gods(Disk 1 of 2).adf
/Amiga/Gods/Gods(Disk 2 of 2).adf

EmulationStation shows the Gods folder with a star on the left because it's a folder with multiple files. BUT the the star is also used to represent Favorites!

I have a ton of multi-disk amiga games (was common) and all have stars.
I have no idea of those entries which ones are Folders or Favorites.

Thanks

[image]
[image]

christianhaitian, 2024-04-25:
I'll review this and see what's involved to accommodate that change.

christianhaitian, 2024-04-25:
This has been accommodated with the update released today.  Folders are now designated with a right arrow.

closed 2024-04-25 by christianhaitian