christianhaitian/arkos #1046: Bugs RGB30 can't turn off. Always restart.

Awakedewe, 2024-04-26:
My Powkiddy rgb30 after new update can't turn off, always restart. Thanks🙏

christianhaitian, 2024-04-26:
Try replacing the existing rk3566-OC.dtb file in the fat32 BOOT partition with the one inside the attached zip file. You can do this with the sd card inserted in your computer. Then try booting ArkOS with the sd card and see if it resolves this issue.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/15129737/rk3566-OC.zip)

jesserankin, 2024-04-26:
This worked for me on a v2 RGB30. Thanks!

Nightchanger, 2024-04-28:
Tried the fix there, it seems to not work when I turned on the "switch power button tap to off". Can only hard turn off and use quit>Shutdown System

christianhaitian, 2024-04-29:
> Tried the fix there, it seems to not work when I turned on the "switch power button tap to off". Can only hard turn off and use quit>Shutdown System

That setting would not impact if the system reboots after shutdown.  Especially if you're able to do Quit>Shutdown system and the unit stays off and not reboot.  Are you sure you were not putting the system to sleep instead of powering off when you had that feature on?

christianhaitian, 2024-05-02:
Try opening your unit and unplug the battery.

denizonm, 2024-05-03:
I have stopped being able to boot after replacing the file, it gets stuck on the ArkOS logo. Is there an easy way to get the same original file?

christianhaitian, 2024-05-04:
Just grab it from the latest ArkOS image using 7zip

Jesquivel2908, 2024-05-09:
I'm having the same issue after installing the latest version of ArkOs on my RGB30. If I select "quit" from the settings menu the device just restarts, I have to press and hold the off button to turn the device off.

Scoonysaurus, 2024-05-10:
For anyone that used Rocknix or Jelos prior to using ARC like me and is having this issue, try putting in the rocknix sd card, booting up and powering off through quit. Then put your ARC card back in. Haven't had the issue since I did that. I read on reddit that the reason for this issue is that "it hijacks the power behavior in a way that causes this "reset" action rather than a shutdown".

Jesquivel2908, 2024-05-10:
> For anyone that used Rocknix or Jelos prior to using ARC like me and is having this issue, try putting in the rocknix sd card, booting up and powering off through quit. Then put your ARC card back in. Haven't had the issue since I did that. I read on reddit that the reason for this issue is that Rocknix "hijacks the power behavior in a way that causes this "reset" action rather than a shutdown".

Now I have the same issue while using the Jelos SD card😫 it just won't power off from the menu

Scoonysaurus, 2024-05-10:
> Now I have the same issue while using the Jelos SD card😫 it just won't power off from the menu

I think Jelos may cause the issue also. I would download the most recent version of Rocknix and use that to boot up and follow the directions above. Then I'd only use ARK from then on.

JesusArmy, 2024-05-22:
For info I just had same thing with my RG353PS. Not sure if Rocknix is really the guilty one, since the timeline is that I did tried Rocknix like two weeks ago with no issue. Then I have used my regular ArkOS a dozen time. And yesterday I wanted to start fresh with ArkOS with a one SDcard setup to leave the second one available for medias. I have then imaged the last ArkOS to an empty SD card, went through the setup until the end, and after system was finalized I could not stop it from menu but only by long pressing power button. I switched back to my previous ArkOS SD card and could not shutdown as well. I landed here and some other online posts, did try to boot on the Rocknix SD card, stop properly and now my new ArkOS is ok (for now).

JesusArmy, 2024-05-22:
Have started Kodi on that "fresh new ArkOS install" and ended up with the same problem again... :(

christianhaitian, 2024-05-22:
Try this https://github.com/christianhaitian/arkos/issues/1046#issuecomment-2089311455

JesusArmy, 2024-05-23:
Thanks for the suggestion. I will check for some tutos to see if disassembling the RG353PS case is just about unscrewing or if more difficult with prying stuff like on typical Sony, Microsoft and Nintendo's systems, which make it more tricky to open without breaking any small plastic part.
I may fall back to my good old JelOS card for now until I fix that permanently for when using ArkOS.

JesusArmy, 2024-05-23:
I wonder if not related to the single or dual sd setup. Looks like my setup with ArkOS and dual sd card has no problem. But it happen quickly once I try the new ArkOS SDcard that I did in single card mode...

patroldo, 2024-06-05:
Hi. I think problem can be reproduced if shutdown happense unexpectedly, not via "Start > Quit > Shutdown system"
Battery dead while gaming

Shutting down system with rocknix is helping in my case

Genomans, 2024-06-08:
don't know why but this became a problem as well for me on my rg353VS it won't shutdown but always restart

dsniper1001, 2024-06-09:
this is still happening, I don't know why this was marked as closed....

Genomans, 2024-06-10:
Okay found the problem it tuns out that manually restarting the system by pressing start/quit/restart system does indeed cause this issue

czapa86, 2024-06-10:
yes your are right, i can replicate the issue by pressing start/quit/restart system

christianhaitian, 2024-06-10:
Just tested this on my end and shutdown still works fine for me.

christianhaitian, 2024-06-10:
On my rgb30v2, it is a fresh install i created yesterday and the problem doesn’t happen on my end.

dsniper1001, 2024-06-10:
> On my rgb30v2, it is a fresh install i created yesterday and the problem doesn’t happen on my end.

I see. The problem in my case at least , is with the RGB20SX. No idea if it's exactly the same HW in the RGB30 , but merely reportying this happened. I do not have this issue in ROCKNIX.

christianhaitian, 2024-06-10:
Same for a rg503 which is of the same chipset.  Did you test ROCkNIX at any time on your unit?  If so, have you tried this? https://github.com/christianhaitian/arkos/issues/1046#issuecomment-2103928667

dsniper1001, 2024-06-10:
> Same for a rg503 which is of the same chipset. Did you test ROCkNIX at any time on your unit? If so, have you tried this? [#1046 (comment)](https://github.com/christianhaitian/arkos/issues/1046#issuecomment-2103928667)

Another weird thing is what booting on ROCKNIX fixes the issue temporarily. But still makes me have to have an SD with rocknix just to keep using arkos regularly. I did not let the battery die, I did not forcedfully turn it off. This happened on the first boot and putting some roms to test if it worked.

christianhaitian, 2024-06-10:
Then it sounds like maybe a battery pull would be necessary. Otherwise, i have no idea at the moment on how to resolve the issue especially since I can’t replicate it yet.

dsniper1001, 2024-06-10:
> Then it sounds like maybe a battery pull would be necessary. Otherwise, i have no idea at the moment on how to resolve the issue especially since I can’t replicate it yet.

I rather not open the device. And booting on ROCKNIX fixes it, so it's some sort of internal bios state that is saved or soemthing that I'm not aware of as I don't know the architecture of these consoles.

czapa86, 2024-06-10:
I'm using ROCKNIX for shutdown the system correctly then changing card to ArkOS and doing restart via start/quit/restart system. Can you try using ROCKNIX on your device and then try restart in ArkOS? Probably the issue is caused by mix those two systems.

Genomans, 2024-06-10:
My RG353vs has this problem as well i even made a clean install of the cfw and updating to the latest patch just i while ago without touching ROCKNIX and it still has this problem, it only affects it when manually restarting the system using start/quit/restart system

czapa86, 2024-06-10:
My guess is that ROCKNIX is leaving some code somewhere on the device so even you made clean install of ArkOS but you used ROCKNIX before you are affected anyway. Never had that issue before but i want to try ROCKNIX and i'm affected by it now.

christianhaitian, 2024-06-10:
I’ll try ROCKNIX on my rgb30v2 to see if i can replicate the issue.

christianhaitian, 2024-06-10:
So after booting the latest image available for ROCKNIX, I can confirm that I now have the same issue of not being able to shutdown in ArkOS after doing a system restart in ArkOS.  Even doing a force power down of the device doesn't resolve the issue.  The only resolve I've been able to find was to open the device and disconnect the battery for 10 seconds and plug it back in to resolve the issue.  Only then was I able to shutdown consistently whether I do a system restart first or not.

Very strange.

dsniper1001, 2024-06-10:
> So after booting the latest image available for ROCKNIX, I can confirm that I now have the same issue of not being able to shutdown in ArkOS after doing a system restart in ArkOS. Even doing a force [...]

I appreciate very much that you took the time to reproduce. 

Thank you Christian. Atm I just muddle through when this happens launching rocknix, but I'm guessing there's some sort of flash storage on that rom that does this. But my guess is as good as any.

christianhaitian, 2024-06-10:
yea in talking with one of the ROCKNIX devs, it seems like it's related with the state that the pmic (power management integrated circuit) that the mainline kernel leaves that's not compatible with the BSP (Board Support Pacakage) kernel used in ArkOS doesn't account for.  Unless it's something that eventually improves with the natural progression of the mainline kernel for this chipset, I doubt there will be a better fix other than the battery pull unfortunately.  I don't have plans to migrate ArkOS to the mainline kernel as it functions well enough as is for the intended gaming needs from my perspective.

dsniper1001, 2024-06-10:
I understand. Thanks for hearing me out at least and see that this is an issue.

angeluskun, 2024-06-23:
> yea in talking with one of the ROCKNIX devs, it seems like it's related with the state that the pmic (power management integrated circuit) that the mainline kernel leaves that's not compatible with [...]

yeah thats exactly thst happens to me on my 353vs hahah i have a rocknix sd just in case because idk what screwdriver i need to open my 353vs

angeluskun, 2024-06-23:
and when i mention this on rocknix discord i was muted HAHAHAH

dsniper1001, 2024-07-03:
Sorry but this bug forces me to need 2 sd cards (one for ArkOS and one for Rocknix) for 1 console as this issue pops up over and over again. I really like ArkOS as I use it across the board in another console I have (RG353V).

I really hope it fixed some day so I can come back.

sword23, 2024-07-16:
@angeluskun If Can't solve the problem, then silence the person who raises the issue.

czapa86, 2024-08-24:
Confirmed unpluging battery solves the issue 100%

mrlehmann88, 2024-09-06:
I've been having this issue, I just remove SD card and hold down power button until it turns off :)

dani7959, 2025-02-02:
Is this being addressed by the recent kernel changes in 01312025?

christianhaitian, 2025-02-02:
No.  The best current method to resolve this issue if coming from a mainline kernel OS like ROCKNIX is to open the unit and unplug battery for a few seconds.

fridayschild, 2025-04-27:
I made an account just to comment that unplugging the battery worked for me. I'm so happy that I found this solution. Especially since my power setting is to shutdown when I press the power button. I can finally use that setting.