christianhaitian/arkos #1047: Sugestion: Audio

lauren7ino, 2024-04-27:
Hi
Just tested your "Fix audio" tool.

Two suggestions:

1) "Fix audio" should reset the volume to 50% or so, in case the user didn't realize they have the volume at 0%. "Fix audio" keeps the current volume.

2) In the end of the "Fix audio", an audio sample could be played (by pressing 'A'), so the user could test that the audio is actually fixed, and not an emulator issue.

Thanks

christianhaitian, 2024-04-28:
Good suggestions.  I've incorporated the suggestions.  If you could help me test this and provide feedback, it would be helpful.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot the device.

4. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then select Fix Audio., hit A.

I'll await your feedback.  Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/15140399/arkosbackup.tar.gz)

lauren7ino, 2024-04-28:
5 stars! Perfect!👌🏼

christianhaitian, 2024-05-25:
This has been included with today's update.

closed 2024-05-25 by christianhaitian

KyleXy101, 2025-03-18:
i tried this on my r36s clone and it didnt work, is there a way you could make one for clones?

christianhaitian, 2025-03-18:
rk3326 devices don’t need this fix. R36 units are not supported either

yoyo81-9, 2025-06-13:
> To fix the sound it is the rd3536k4ka.dtb file that is needed.
> 
> (I went to many many forum), but it was mentioned in this post:
> 
> https://www.reddit.com/r/R36S/comments/1hhn15r/arkos_for_k36_and_r36s_clones/
> 
> scroll down, the post of _manster_
> 
> Take the [#3](https://github.com/christianhaitian/arkos/issues/3) file to fix the sound

Bro I did all this stuff, but I’m stuck in the part of this especific file, where I found it? Where I found the rd3536k4ka.dtb ??

globeron, 2025-06-15:
> 
> Bro I did all this stuff, but I’m stuck in the part of this especific file, where I found it? Where I found the rd3536k4ka.dtb ??
>

In the weblink above go through all the >>comments<< in the link until you see 
a post by someone called "manster"  with all the different dtb files I've found for this image:
(once you have the file then overwrite it on your sd-card and do a reboot of the K36.

Search for Manster post with these items in it.
[rf3536k4ka.dtb #1]
[rf3536k4ka.dtb #2]
**[rf3536k4ka.dtb #3](use this if you have no sound)**
[rf3536k4ka.dtb #4] (inside "CLONE boot partition.zip")

OphelieTxr, 2025-07-23:
Hi, just install this https://drive.google.com/drive/folders/1kOStwUgtx6K1sU3b7zmU1zpicL8Xp9sX for 

[arkos dtbs-20250723T225412Z-1-001.zip](https://github.com/user-attachments/files/21396217/arkos.dtbs-20250723T225412Z-1-001.zip)

sound fix (just use the one with the name of your dtb file)