christianhaitian/arkos #1049: RGB30 can't write to SD2

td0, 2024-04-29:
today I updated arkos on my Powkiddy RGB30, changed the default password, and reused the SD2 from JELOS (FAT32 formatted) by moving the SD2 content from /roms/* dir to the root dir in the SD Card. It was running fine until I let my RGB30 idle (screensaver kicking up). Every time I try to do the savestate, it stuck at 0%, and I can't quit Retroarch either. this behavior persists across system reboot (after I pressed the RGB30 reset button, the savestate still stuck at 0%).

To check what happens, I tried to create an arbitrarily named folder through File Manager, but the new folder didn't get created. Then I tried to verify through ssh by entering the following command:
```
% touch a
touch: cannot touch 'a': Read-only file system
```
it only occurs in /roms2, I can still write in any dirs within the SD1

What I can do at the moment is to toggle "Switch to main SD for Roms", and then back to "Switch to SD2 for Roms".

christianhaitian, 2024-04-29:
Bad sd card

closed 2024-04-29 by christianhaitian

td0, 2024-04-29:
nah man, it was running all fine. I've been using JELOS for about 2 months and never found any issue such as this.
I also used this same SD card for my Nintendo Switch for years and years until I upgraded to higher capacity. So highly doubt that this is an SD card fault. the FAT32 format might be the issue though

christianhaitian, 2024-04-29:
sd cards do go back and one of the first symptoms of them failing is going into read only mode.  Hopefully formatting to exfat resolves it for you if it's not the sd card failing.

td0, 2024-04-29:
What I've just tried to do was to set the password back to 'ark', let the handheld idle until the screensaver kicked up, and now the savestates works normally.

If the issue comes up again, I'll try to re-format it into exfat & update here. thanks anyway