christianhaitian/arkos #1052: [Feature Request] Rotating logo to make every boot fresh

ekoprihastomo, 2024-05-01:
First massive thanks for your work developing ArkOS 🙇‍♂️

I know I can replace logo.bmp with my own pic
My suggestion is how about a folder where I can store multiple pictures for logo with fixed filename like logo001.bmp, logo002.bmp and so on
At the shutdown process add simple copy and rename from logo folder to boot partition
The order is up to you, can be random & sequential
By doing it at the shutdown process, boot process will still the same and we'll see surprise logo every time we turn on our device

Just my random idea 😁
Again massive thanks for your work 🙇‍♂️❤

christianhaitian, 2024-05-01:
This is already possible on rk3326 devices and was enabled at one point but most people seem to not like the feature and I got a request to disable it.  You can reenable it by doing the following:

- You will need a network connection in order to remote into your installation.
- If you need some info on how to ssh into your device, follow the instructions here: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-do-i-ssh-into-ArkOS and once you're in a ssh session, copy and paste the following command then press enter:
`echo "@reboot /home/ark/.config/imageshift.sh &" | sudo tee -a /var/spool/cron/crontabs/root`
- Copy all your logos into a BMPs (case sensitive) folder on your fat32 boot partition then it should change to a different logo on every boot.

ekoprihastomo, 2024-05-01:
Wait what???
People actually asked you to disable additional awesome feature? why?? 😱
I have no idea about this and when it's happened but I assume if I don't want the logo to change automatically I can just use one logo right?
I also assume that it won't have performance hit to gaming right?
Then why people asked you to disable it?? can't think of any reason 🤔

Btw I understand the folder but is there any specific filename for the logos?

closed 2024-05-02 by christianhaitian

ekoprihastomo, 2024-05-03:
Tried your image shift command and succeeded 😍🎉
Thank you master for the info 🙇‍♂️

I understand if you kinda upset with this matter due to the complaints you've received back then and won't enable this feature
I also understand I'm not in position to ask you anything here but I beg you please don't remove this in your future update 🙇‍♂️

Thank you soooooo much master 🍻🙇‍♂️

christianhaitian, 2024-05-03:
Happy it's working for you.  I won't remove it in any future updates.  It will just remain disabled on initial setup from a fresh image.  Enjoy!