christianhaitian/arkos #1053: Portmaster: Bleed 1 not working (cannot open assembly "MONOMODDED_Bleed.exe" no such file or directory)

paletochen, 2024-05-01:
### Describe the Issue (If applicable)

I purchased Bleed 1 and Bleed 2 in order to use them in portmaster.
I downloaded the linux version.
While Bleed 2 runs in Portmaster, Bleed 1 doesn't.

Error thrown:
**cannot open assembly "MONOMODDED_Bleed.exe" no such file or directory**

### How can the issue be reproduced? (If applicable)
Download the Linux version of Bleed 1.
Copy content in ports/bleed/gamedata/

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
RG353V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS 2.0 (04242024)

### Anything else you'd like to include that may help to review this issue or feature request?

paletochen, 2024-05-01:
I just tried with the macos version, the dmg file and the port works. so it seems there is an issue with the Linux version

christianhaitian, 2024-05-01:
Hi, I'm no longer involved with PortMaster.  There's a separate group that handles all of that.  You'll have to report the issue to the PortMaster group for further assistance. See here for more info on how to seek help with that. https://portmaster.games/

closed 2024-05-01 by christianhaitian