christianhaitian/arkos #1054: Fail to mount 2nd sdcard - Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)

rabin-io, 2024-05-02:
### The Issue

When trying to use 2nd SDCARD slot, I get the message 

    Could not find a Fat, Fat32, Exfat, EXT4, or NTFS based sdcard to mount from slot2

When I log into the device, and watch dmesg i see this messages when I pop-in the SDCARD

```
[ 1222.539861] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[ 1222.903331] net_ratelimit: 22 callbacks suppressed
[ 1223.293661] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1223.299553] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1223.347146] mmc1: error -110 whilst initialising SD card
[ 1223.855616] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1223.861485] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1223.883922] mmc_host mmc1: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[ 1224.411636] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1224.417512] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1224.439887] mmc_host mmc1: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[ 1224.967649] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1224.973523] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1224.995910] mmc_host mmc1: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[ 1225.535675] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1225.541572] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
```

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

    R36S

### What version of ArkOS are you on?

   v2.0 (04/24/2024)

christianhaitian, 2024-05-02:
You have either a bad sd card, a sd card that's not compatible with the device or the 2nd slot is bad on your unit.  I don't support the R36s units.  You can go here for more info on support. https://aeolusux.github.io/ArkOS-R3XS/

closed 2024-05-02 by christianhaitian

rabin-io, 2024-05-02:
Hi @christianhaitian thank you for responding quickly, I know you not support the R36S, but I think it might be a kernel related issue, with setting the speed on the port. I don't think the issue is with the SD card itself, as I was able to verify it with f3. It might be the size and spec maybe (it's Sandisk Extreme Pro 128GB / U3 / A2) with a chapter card it dose seems to work, but I do see this messges on insert.

```
[  419.406620] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[  419.491458] mmc_host mmc1: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[  419.714340] dwmmc_rockchip ff380000.dwmmc: Successfully tuned phase to 168
[  419.714387] mmc1: new ultra high speed SDR104 SDXC card at address 0001
[  419.715388] mmcblk1: mmc1:0001 SD 58.2 GiB 
[  419.717086]  mmcblk1: p3
[  426.338893] mmcblk0: error -110 sending status command, retrying
[  426.345019] mmcblk0: error -110 sending status command, retrying
[  426.351137] mmcblk0: error -110 sending status command, aborting
[  426.374541] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[  426.637363] mmc_host mmc0: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[  426.773107] dwmmc_rockchip ff370000.dwmmc: Successfully tuned phase to 210
[  426.773135] mmc0: tried to reset card
```

So if you think there is nothing else to do here, you can keep the issue closed.

Thank for all your efforts.

christianhaitian, 2024-05-03:
The R3x devices are known to have issues with quality sd cards especially with the 2nd sd card slot.  Devices such as the RG351MP and other higher quality devices don't have these issues using the same kernel.

rabin-io, 2024-05-04:
Is there an option to set the port speed, [like with the RPi](https://forums.raspberrypi.com/viewtopic.php?t=351894)?

christianhaitian, 2024-05-04:
Maybe in the kernel.  Not sure from boot.ini.  Kernel source is here:< https://github.com/christianhaitian/linux/tree/rg351 >