christianhaitian/arkos #1057: Bluetooth Xbox Series Controller

czapa86, 2024-05-05:
Hello,

i'm trying to connect and use xbox wireless controller on my RG353M device. I had no problem with pairing it however bindings in games are wrong. LB is always select and RB start, there are also missing buttons. How to configure it properly? Back button is hotkey but I'm not able to open retroarch menu as start doesn't do nothing. When i configure controllers in emulationstation menu it doesn't change anything. Also i was trying to configure in retroarch launching it from menu but also there is no change in games. Pls help.

christianhaitian, 2024-05-05:
Hello,  

See here: https://github.com/christianhaitian/arkos/issues/665

closed 2024-05-05 by christianhaitian

MeneerJansen, 2024-09-28:
With all due respect but this does not "resolve" the bug.

czapa86, 2024-09-28:
Check
https://github.com/christianhaitian/arkos/issues/1064