christianhaitian/arkos #1059: Sugestion: Alpha sorting: Per Pixel for flycast emulator on rg353Vs

Genomans, 2024-05-06:
do you think its possible to have Per pixel on Alpha shorting for the flycast emulator core becasue games like melty blood have texture problems

christianhaitian, 2024-05-06:
I don't think it's possible with gles only support but I'm going to try patching it in and see if it's selectable and if it works.

christianhaitian, 2024-05-07:
Well I was able to enable it and set it in core options.  Games seem to play but I don't know if it will make a difference as I didn't try games like melty blood that may benefit from it.  Please do a core update in retroarch and retroarch32 and try the option and let me know if you notice any changes.

Thanks,

Genomans, 2024-05-07:
Thanks will check it out later. Will give any feedback.

christianhaitian, 2024-05-25:
Closing as i've done what I know on how to enable this feature at this time.

closed 2024-05-25 by christianhaitian