christianhaitian/arkos #1060: Bug: Upgraded to 04242024 but want to reboot to second sd card with 01272024 but won't boot

riedyw, 2024-05-06:
I recently received an Anbernic RG353V in March 2024. I successfully downloaded the 01272024 image from your website. I pushed image to SD card. I was able to successfully boot with 01272024 and subsequently upgraded to 04242024. I transferred my game ROMs to a second SD card and was able to successfully load and play games from the secondary card.

I subsequently was mucking around with resetting default configs and now things are not working properly:

1. WonderSwan and WonderSwan Color games are no longer appearing in the list of consoles whereas they were appearing before
2. Going into Options menu there is no longer a Tools menu so ThemeMaster and PortMaster are no longer visible. From another FAQ I read I took note of where ThemeMaster should be and it's on the primary SD card in the right location
3. The scrolling mechanism is behaving very differently. R1/L1 go to next/previous console (respectively) whereas before it was pagedown/pageup in the list of games and that is the behavior I want. I can't find where I made those changes so I don't know how to change it back.

I am NOT looking for you to help me fix my weird config as I got myself there. My plan was to image a new 16GB SD card with the 01272024 image and start over.

The problem is that when I insert the new 01272024 SD card into the RG353V first slot the unit gets stuck on the ArkOS boot screen. It never gets to the dots loading screen where it displays the version of the image. I wasn't impatient as I waited well over 30 minutes for the system to load as when I did it the first time the initial boot was lengthy while it resized partitions and ran a bunch of scripts.

I'd like to be able to boot another SD card with an earlier release of ArkOS both now because I mucked things up, and also in the future if I muck things up again.

I have reviewed the FAQ and did not see anything resembling this issue.

I've read through the open issues and several hundred of the closed issues and didn't see anything resembling my issue.

riedyw, 2024-05-06:
Wow - thanks for the rapid reply. I don't think your suggestions above are the issue.

I was rereading the closed issues and read #837 and that rung some bells for me. I **may** have done that originally but my memory is not perfect. I will attempt to do that tonight when I get home from work. If that is indeed the cause of the issue I will add another note here.

christianhaitian, 2024-05-06:
If you were able to consistently boot ArkOS without issue before making some changes then that issue in 837 is different from what you describe here.

riedyw, 2024-05-06:
I was describing the situation that prompted me to want to boot an older OS. I was NOT looking for assistance in fixing my configuration screwup. I was just looking for solution to be able to boot older OS. Sorry for my rambling.

christianhaitian, 2024-05-06:
Oh I see.  Well you were able to load the latest image fine before.  So what's different now?  Are you using a different brand sd card now?

riedyw, 2024-05-07:
Issue 837. I apparently forgot I did that originally. Please ose this issue

riedyw, 2024-05-07:
It was issue 837. I forgot I had done it previously

closed 2024-05-07 by christianhaitian